jzhzhang/ROSEFusion #1: Frame could not be processed

Mr-Kyrie, 2021-07-16:
I downloaded the seq file and related configuration files,but it could run the programe with a small size synthesis seq file and related configuration files in github. The erro information shows "Frame could not be processed" and every frame all like this,so I don't know how to fit the bug.

jzhzhang, 2021-07-16:
@Mr-Kyrie  Please share more details, like the .seq and related configuration files you used.  
As a quick suggestion, you can try the [example](https://drive.google.com/drive/folders/1vW5GV2xsJN1kIrl-5JZX1fUrpjCtp5AS?usp=sharing)  to make sure your program is installed correctly.

Mr-Kyrie, 2021-07-16:
I just use the link of example about .seq and related configuration files and run the ./ROSEFsuion with you config file and .seq.However, only depth map and color map are displayed in the running process, and the number of vertex of the point cloud is 0. The output information of the command line is "frame is not processed".So I don't know how to solve it.Please give me some advice,thanks.

jzhzhang, 2021-07-16:
Do you check the version of CUDA and OpenCV? 
- OpenCV with CUDA ( v4.5.0)
- CUDA (11.0 , 11.1) 
I only test the code with rtx 2080 SUPER and rtx 3080 in ubuntu 16.04.

Mr-Kyrie, 2021-07-16:
I use OpenCV with CUDA(v4.5.0) and not with sfm module and python 2 and 3,because it always occure erros.
CUDA is 11.0,code with 1080.
May I add your wechat for facilitateing follow-up communication?My wechat is lw574970565.

closed 2021-07-21 by Mr-Kyrie

rlczddl, 2021-12-14:
@Mr-Kyrie how to solve? i meet the same problem.

renhaofan, 2022-01-01:
I meet the same problem with Docker on example dataset