jzhzhang/ROSEFusion #3: Segmentation fault during fusion

Kyrie-Irving07, 2021-10-04:
Hi, 
I followed the instructions to run the fusion example without any error in the provided docker. 
However, when testing on my own data,  it appeared "Segmentation fault: (core dumped)" after printing "Read seq file: XXXX".

The '.seq' file was made successfully through './seq_gen' with my .yaml file:
small_room_seq.yaml
%YAML:1.0
association_file: /share/small_room/rose_data//associations.txt
color_to_depth: 1
depth_ratio: 1000
img_height: 1440
img_width: 1920
output_path: /share/small_room/rose_data/
seq_name: small_room
source_path: /share/small_room/rose_data/

the controller.yaml file is similar to that of the example:
small_room_controller.yaml
%YAML:1.0
PST_path: ../PST/
init_fitness: 0.5
max_iteration: 20
momentum: 0.9
render_surface: 0
save_scene: 1
save_trajectory: 1
scaling_coefficient1: 0.18
scaling_coefficient2: 0.09
scaling_inherit_directly: 0

There are data and camera config files:
small_room_data.yaml
%YAML:1.0
name: small_room
result_path: /share/small_room/rose_data//result/
seq_path: /share/small_room/rose_data//small_room.seq
truncated_size: 60
voxel_size: 20.0

small_room_camera.yaml
%YAML:1.0
Camera.cx: 947.3194580078125
Camera.cy: 722.2567138671875
Camera.fx: 1599.2437744140625
Camera.fy: 1599.2437744140625
Camera.height: 1440
Camera.width: 1920

It turns out that something wrong with my usage. Logs as follows:
Read configure file                                                                                                     
Init configuration 
/share/small_room/rose_data//small_room_camera.yaml                                                                                                      
/share/small_room/rose_data//small_room_data.yaml 
/share/small_room/rose_data//small_room_controller.yaml  
Read seq file: /share/small_room/rose_data//small_room.seq     
Segmentation fault (core dumped)         

Thank you!

jzhzhang, 2021-10-04:
I'm not sure if the "/" in your path ```/share/small_room/rose_data//``` may be the problem. And it seems you didn't give the parameters of the volume. You can make a volume like the [FastCaMo-real data](https://github.com/jzhzhang/ROSEFusion/blob/main/dataset_config/fastCaMo-Real/apartment_1.yaml#L8-L13)

Kyrie-Irving07, 2021-10-04:
Problem solved by giving volume parameters. Thanks a lot!

closed 2021-10-04 by Kyrie-Irving07