jzhzhang/ROSEFusion #5: Visualization issues

Darker-zq, 2021-11-18:
I downloaded the image that you uploaded in docker. And downloaded the small size synthesis example than you uploaded. But there was no visual interface when i run ./ROSEFsuion  example_camera_config.yaml example_data_config.yaml example_controller_config.yaml. The program did not crash and 'example.txt' generated successfully .
What caused the visual interface not to appear?Whan should i do?
thanks!
[image]

abc619539082, 2021-11-18:
you should check your example_data_config.yaml to change the result path to yours

Darker-zq, 2021-11-18:
thanks your reply.
the new issues was occured.when i run it again.
[image]
I'm sorry to disturb you.

jzhzhang, 2021-11-19:
@Darker-zq  Thanks for your interest and @abc619539082  thanks for your kind help.

The visual interface (rendering surface) is disabled because the docker is not designed for the GUI output. If you want to visualize the GUI of ROSEFusion in docker, please try the following steps:
1. Change the "render_surface" of controller.yaml to "1"
2. Bind the GUI output to some program.  Here is a quick [reference](https://stackoverflow.com/questions/16296753/can-you-run-gui-applications-in-a-linux-docker-container).

Darker-zq, 2021-11-19:
thank yours answer。 Why the frame cloud not be processed? The .ply file is not have element vertex.
[image]

jzhzhang, 2021-11-19:
This may relate to some of your changes to configuration files. Have you tried to directly use the ```run_example.sh``` or ```run_stairwell.sh``` in ```/home/code/ROSEFusion-main/build```?  These are prepared scripts to run the ROSEFusion.

Darker-zq, 2021-11-19:
yes, I had been ran the run_example.sh and  run_stairwell.sh. And the results were still the same.

jzhzhang, 2021-11-19:
Can you give me more details about your hardware and modifications?  And did the trajectory results are normal?

Darker-zq, 2021-11-19:
I downloaded the docker file and did not make any changes. I checked the path in the configuration file and it was consistent. At the same time, the point cloud was still not processed after trying to run the .sh file.
The host computer used ubuntu18.04 system and the GPU is nvidia  GeForce RTX 3090.
The  trajectory results are still incorrect. The example.txt and stairwell.txt only have [0 0 0 0 0 1].

zhanghua7099, 2021-11-25:
https://github.com/jzhzhang/ROSEFusion/issues/5#issuecomment-973698135

Same problems.

I didn't use ROSEFusion via Docker, but occur the same errors. The trajectory and mesh file were not saved properly.
My GPU is RTX 3090. 
```
Frame could not be processed
n:2
Frame could not be processed
n:3
Frame could not be processed
n:4
Frame could not be processed
n:5
Frame could not be processed
n:6
Frame could not be processed
n:7
```

Darker-zq, 2021-11-25:
@zhanghua7099  please check the architecture in the L22 of /home/code/ROSEFusion-main/CMakeList.txt is compatible with your own graphics card. If not, change the sm_75 and compute_75 to sm_86 and compute_86, then rebuild the ROSEFusion.
It is worth noting that this is just a suggestion of mine, I did this in docker but still did not generate trajectory and point cloud。

jcyhcs, 2021-11-26:
@Darker-zq  @jzhzhang 
i also got the same error with you,which said "Frame could not be processed",oh,what's the problem?
please help me!

jzhzhang, 2021-11-26:
@jcyhcs @Darker-zq @zhanghua7099  Sorry for all the trouble. 

I have tested the code on 2080 SUPER, 3080Ti and Quadro T1000 without reproducing the same bug. I will further check the code and the environment.

jcyhcs, 2021-11-26:
@jzhzhang  
thx, then my enviroment is:
Quadro P4000: Driver Version: 460.91.03    CUDA Version: 11.2
opencv:4.5.1
this problem boring me,so please help!

Darker-zq, 2021-11-26:
for the problem of "Frame could not be processed", please check the architecture in the L22 of /home/code/ROSEFusion-main/CMakeList.txt is compatible with your own graphics card. If not, change the sm_75 and compute_75 to sm_xx and compute_xx, then rebuild the ROSEFusion. The cuda arch(value of xx)could Inquired on [https://developer.nvidia.com/cuda-gpus](https://developer.nvidia.com/cuda-gpus)

jzhzhang, 2021-11-26:
@jcyhcs @Darker-zq  Great！Thanks for your interest in our project.

closed 2021-11-26 by jzhzhang