jzhzhang/ROSEFusion #7: 3D reconstruction process visualization issue

Goughhhhhh, 2021-12-07:
There was no **3D reconstruction process visual interface**(the interface as shown on the right side of the figure) when I run ./ROSEFsuion example_camera_config.yaml example_data_config.yaml example_controller_config.yaml. But the color image, the depth image and shaded image(the interfaces as shown on the left side of the figure) are displayed, and *.txt and *.ply are generated successfully .
How can I display the **3D reconstruction process visual interface**? What should I do? Thanks!
[image]

jzhzhang, 2021-12-07:
The full "3D reconstruction process visual interface" is not integrated into the open-source version.  It's intended that only RGB, depth and shaded images are displayed, which reduces the complexity of the system.

Goughhhhhh, 2021-12-07:
> The full "3D reconstruction process visual interface" is not integrated into the open-source version. It's intended that only RGB, depth and shaded images are displayed, which reduces the complexity of the system.

closed 2021-12-07 by Goughhhhhh

Goughhhhhh, 2021-12-07:
Thanks for your reply! 
Could you tell me how can I do for display this process interface?
Thanks.

jzhzhang, 2021-12-07:
@Goughhhhhh  You can build your own shader or online transfer (like ROS message) the reconstruction results to some well-built shader.