jzhzhang/ROSEFusion #8: Cannot run example

rlczddl, 2021-12-14:
Hi,my env is: ubuntu18.04、cuda11.1、opencv4.4 with cuda and i can build the ROSEFusion exe on my machine,but I meet the "Frame could not be processed" problem. Here is the config file:
[image]
[image]
[image]

Any suggestions?

jzhzhang, 2021-12-14:
Please check if your CUDA architecture is compatible with the [CMakeLists.txt](https://github.com/jzhzhang/ROSEFusion/blob/02a847994c835bf6381201bd91c755fd2ae4dc76/CMakeLists.txt#L22). 

The previous [issue#5](https://github.com/jzhzhang/ROSEFusion/issues/5#issuecomment-979943151) may relate to your problem.

rlczddl, 2021-12-14:
@jzhzhang OK，I see.

closed 2021-12-14 by rlczddl