jzhzhang/ROSEFusion #9: Data Preparation about  ./seq_gen  <.yaml>

Darker-zq, 2021-12-15:
I'm sorry to disturb you.
An error occurred when I ran ./seq_gen.
I recorded my own data by using Azure Kinect DK.but these data cannot be converted into .seq format files.
The error is as follows：
[image]

My data as follows:

[image]

**color**

[image]

**depth**

[image]

**sequence_information.yaml**

[image]

**associations.txt**
[image]

I could convert the data that you provided to .seq format successfully .
How can I solve it, thank you very much.

jzhzhang, 2021-12-15:
Please check your depth maps and make sure they are stored as 16-bit integers.

Darker-zq, 2021-12-17:
thanks for your answer.
The actual problem is that I made a basic mistake that my sequence_information.yaml(402test_2.yaml)  starts with a number.

closed 2021-12-17 by Darker-zq