jzhzhang/ROSEFusion #10: Frame could not be processed

cybaol, 2022-01-10:
when I run “ ./build/RoseFusion ./camera_config/FastCaMo_camera.yaml ./dataset_config/fastCaMo-Synth/raw/apartment_1_raw.yaml ./controller_config/controller_terminal.yaml ”

it mentioned that Frame could not be processed.

Here is my config:

FastCaMo_camera.yaml:
Camera.fx: 320.0
Camera.fy: 320.0
Camera.cx: 319.5
Camera.cy: 239.5
Camera.width: 640
Camera.height: 480

apartment_1_raw.yaml
name: apartment_1_raw
# seq_path: /media/jiazhao/Jiazhao_BACKUP/FastCaMo-synth/synthesis_seq/apartment_1/apartment_1_raw.seq
# result_path: /media/jiazhao/Jiazhao_BACKUP/FastCaMo-synth/FastCaMo-Synth-result/
seq_path: /home/kino/mnt/FastCaMo-synth-data_raw/apartment_1/apartment_1.seq
result_path: /home/kino/project/3D-Reconstruction/ROSEFusion/result/
voxel_size: 20
truncated_size: 60
voxel_x: 712
voxel_y: 712
voxel_z: 712
init_x: 0
init_y: 0
init_z: 1000

controller_terminal.yaml:
render_surface: 1
PST_path: "/home/kino/project/3D-Reconstruction/ROSEFusion/PST/"
save_trajectory: 1
save_scene: 1
max_iteration: 20
scaling_coefficient1: 0.18
scaling_coefficient2: 0.09
init_fitness: 0.5
momentum: 0.9
scaling_inherit_directly: 0

apartment_1_seq.yaml:
seq_name: apartment_1
# association_file: /media/jiazhao/Jiazhao_BACKUP/FastCaMo-real/FastCaMo-real-data/apartment_1/associations.txt
# output_path: /media/jiazhao/Jiazhao_BACKUP/FastCaMo-real/FastCaMo-real-seq/
# source_path: /media/jiazhao/Jiazhao_BACKUP/FastCaMo-real/FastCaMo-real-data/apartment_1/
association_file: /home/kino/mnt/FastCaMo-synth-data_raw/apartment_1/associations.txt
output_path: /home/kino/mnt/FastCaMo-synth-data_raw/apartment_1/
source_path: /home/kino/mnt/FastCaMo-synth-data_raw/apartment_1/
img_height: 480
img_width: 640
depth_ratio: 1000
color_to_depth: 1

jzhzhang, 2022-01-10:
Thanks for your interest in our project. Did you test your installation with the `example.seq`?

 If you have successfully run the example, please check your .seq is correctly generated. And also please make sure your `path` in config files is correct.

cybaol, 2022-02-13:
> Thanks for your interest in our project. Did you test your installation with the `example.seq`?
> 
> If you have successfully run the example, please check your .seq is correctly generated. And also please make sure your `path` in config files is correct.
I have problem, how to generate my own Particle Swarm Template file.

Goughhhhhh, 2022-02-14:
@jzhzhang Is there any skills of using the parameters like voxel_x, voxel_y, voxel_z, init_x, init_y and init_z? I always have problem "this frame cannot be process" when I run my data. When I change those parameters, things get changed, but it's always blind. Could you give me some suggestions on setting those parameters?

jzhzhang, 2022-02-14:
@Goughhhhhh 
The voxel_x, voxel_y and voxel_z define the length (voxel_x/y/z*voxel_size) of the volume. And the camera pose initially start at the center of the volume with bias defined by the  init_x, init_y and init_z. You can refer to [data_types.h L126-L133](https://github.com/jzhzhang/ROSEFusion/blob/main/include/data_types.h#L126) for more details.

closed 2022-04-15 by jzhzhang

ghy0324, 2022-09-13:
@Goughhhhhh Hi! Have you figured out how to set these parameters?