jzhzhang/ROSEFusion #12: strange display

ronaldpan, 2022-01-15:
ROSEFusion example_camera.yaml example_data.yaml example_controller.yaml
I got this strange display
[image]
What is wrong with it?

jzhzhang, 2022-01-15:
This may be related to the visualization framework. Could you please tell me the version of your Pangolin? 

And is there anything wrong with the results, like trajectory or reconstructed scene?

ronaldpan, 2022-01-15:
I correct the bug as follows:

`void DataReader::readFile()
{
    for (int i=0;i<numFrames;i++){
        char* depth_buffer = new char[height*width*2];
        auto tmp = fread(depth_buffer,height*width*2,1,fp);
        assert(tmp);

        uchar* color_buffer = new uchar[height * width * 3];
        tmp = fread(color_buffer,height*width*3,1,fp);
        assert(tmp);

        v_color.push_back(cv::Mat(height,width,CV_8UC3,color_buffer).clone());
        v_depth.push_back(cv::Mat(height,width,CV_16UC1,depth_buffer).clone());
        delete[]depth_buffer;
        delete[]color_buffer;
    }

}`

jzhzhang, 2022-01-15:
@ronaldpan  Great! Thanks for pointing it out. 

I will review and update the code in the near future.

closed 2022-01-15 by jzhzhang