jzhzhang/ROSEFusion #14: Camera drift after loop closure

ronaldpan, 2022-01-16:
I try a slow speed sequence and find camera drift after loop closure. Is there any method to solve this problem?
[image]

jzhzhang, 2022-01-16:
The ROSEFusion only takes the depth for camera tracking. There is no direct way to solve the loop closure based on ROSEFusion.

Here are some quick tips that may improve the quality of the reconstruction: 
1) you can use a smaller voxel size 
2) using a depth sensor with higher quality depth map.

ronaldpan, 2022-01-16:
I think it is due to no global pose optimization in ROSEFusion.

jzhzhang, 2022-01-16:
The ROSEFusion currently only considers the relative motion between consecutive frames. Admitting RGB images for global optimization might solve this problem as you mentioned.

closed 2022-02-16 by jzhzhang

tangqi1012, 2023-06-14:
Hello, is this your own data set for testing? How do you generate PST file?