jzhzhang/ROSEFusion #16: Do I need to recreate PST for my own data?

pandahops1, 2022-02-17:
In the paper you mentioned:
> PST is a set of particles pre-sampled uniformly within the unit sphere in the 6D space of camera pose. Through moving and rescaling the pre-sampled PST guided by swarm intelligence, our method is able to drive tens of thousands of particles to locate and cover a good local optimum extremely fast and robustly.

But in readme.md you said:

> You can find the .tiff images in [PST dicrectory](https://github.com/jzhzhang/ROSEFusion/blob/main/PST), and please replace the PST path in controller_config.yaml with your own path.

Shoud I create my own PST? Thanks.

pandahops1, 2022-02-18:
#10 ok it seems like its answered here.

closed 2022-02-18 by pandahops1

xiaopeige, 2022-04-29:
How to generate PST, I don't know how to generate，，thank you

jzhzhang, 2022-04-30:
@xiaopeige 
We use the Poisson-disk sampling to generate the PST within a unit sphere. And the particles within the PST range from -1 to 1. 

You can adapt to other uniform sampling strategies for simplification, like FPS or voxel sampling.

xiaopeige, 2022-05-05:
> @xiaopeige We use the Poisson-disk sampling to generate the PST within a unit sphere. And the particles within the PST range from -1 to 1.
> 
> You can adapt to other uniform sampling strategies for simplification, like FPS or voxel sampling.

Sorry, I still don't know how to generate PST, do you have relevant examples or references? thank you

tangqi1012, 2023-06-14:
> Sorry, I still don't know how to generate PST, do you have relevant examples or references? thank you

Hello, do you know how to generate PST  now？