jzhzhang/ROSEFusion #17: use only depth images

pandahops1, 2022-02-22:
is there a way to modify the code so that only depth images are used?
it seems FrameData has both depth and rgb image as input so it seems kinda hard to modify it.
also i tried changing all the depth images to completely black arrays but the result was worse.

jzhzhang, 2022-02-22:
The ROSEFusion takes the RGB images for visualization.  If you only have the depth images, you can use empty cv::Mat for [color_img in main.cpp](https://github.com/jzhzhang/ROSEFusion/blob/main/src/main.cpp#L79).

pandahops1, 2022-02-22:
i changed that place in main to the following:
```
// bool success = pipeline.process_frame(depth_map, color_img,shaded_img);
cv::Mat dummy_img;
bool success = pipeline.process_frame(depth_map, dummy_img,shaded_img);
```
but after ```cmake ..; cmake --build .```, generating the .seq file and running ROSEFusion with parameters that worked before this change, it produced the following error (and information):
```
Read configure file
Init configuration
/home/<username>/fusion/ROSEFusion/camera_config/cam450.yaml
/home/<username>/fusion/ROSEFusion/dataset_config/brokenscan.yaml
/home/<username>/fusion/ROSEFusion/controller_config/controller.yaml
Read seq file: /home/<username>/3DScan/brokenscan.seq
n:0
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.2) /home/<username>/Downloads/opencv-4.5.2/modules/core/src/cuda/gpu_mat.cu:389: error: (-217:Gpu API call) an illegal memory access was encountered in function 'setTo'

[1]    14168 abort (core dumped)  ./ROSEFusion
```
the window showed up for a moment and closed, unlike before where the window showed the real time depth image, rgb image, etc.
i also tried ```cv::Mat1f dummy_img(row,col);``` but saw the same behavior.
help please. thanks.

jzhzhang, 2022-02-22:
You have to make sure your ```dummy_img``` is valid ( 8UC3 ) and has the same solution as the depth map.

closed 2022-02-23 by pandahops1