jzhzhang/ROSEFusion #19: Could you share FastCaMo-Real Ground truth scan in Baidu cloud?

cdb0y511, 2022-03-10:
Hi, @jzhzhang. Thanks for your amazing work.
I notice the ground truth of FastCaMo-Real is missing cause the original scan is too large to upload to google cloud.
Could you share FastCaMo-Real Ground truth scan in Baidu cloud? 
It will help all Chinese users and surely increases the impact of your work.
Thanks,

jzhzhang, 2022-03-10:
Thanks for your interest in our projects. I will upload the ground truth model to Google Drive or Baiduyun disk by the end of this week. 

The original model from the laser scanner is extremely dense and has a large size of nearly 200GB. Therefore I will uniformly downsample the model and it could still perform as a gt model.

jzhzhang, 2022-03-11:
@cdb0y511  I have already uploaded the gt of FastCaMo-real to Google Drive, and you can download the models from this [link](https://drive.google.com/drive/folders/1AXiTpC-UZ0WLhLCqjfKEPl5H40B0NccI?usp=sharing). I utilized the built-in spatial downsample algorithm from cloudcompare, you can check details form [here](https://www.cloudcompare.org/doc/wiki/index.php?title=Edit%5CSubsample).

If you have any question about the gt models, feel free to contact me.

closed 2022-03-12 by jzhzhang

Goughhhhhh, 2022-08-08:
Thanks for your excellent work. The coordinates of the gt of FastCaMo-real is different form the reconstruction model, which results in a misalignment when evaluating its accuracy using cloudcompare, how can I solve this problem, thank you!

jzhzhang, 2022-08-08:
@Goughhhhhh  Hi, thanks for your interest in our project! 

For your question, It is expected that the reconstruction and GT model is not aligned, because the GT model is obtained from the Laser scanner with it own coordinate system.  So to evaluate your reconstruction, you have to apply an alignment algorithm, like ICP, to have the relative transformation between reconstruction and GT.