jzhzhang/ROSEFusion #20: ROSEFusion/src/main.cpp:57:132: error: use of deleted function ‘constexpr pangolin::GlTexture& pangolin::GlTexture::operator=(const pangolin::GlTexture&)’       camera_config.image_width,camera_config.image_height,GL_RGB,false,0,GL_RGB,GL_UNSIGNED_BYTE)

yCOrex, 2022-03-24:
Occurs during make
refers to every `pangolin::GlTexture()`
The error markers all point to the last bracket

jzhzhang, 2022-03-24:
This problem may relate to the version of your Pangolin.  I recommend you to check this [issue](https://github.com/jzhzhang/ROSEFusion/issues/4), and use the [older version](https://github.com/stevenlovegrove/Pangolin/releases/tag/v0.5) of Pangolin.

yCOrex, 2022-03-24:
Ive seen the issue and used v0.5 before,but it does work,so I posted this issue

yCOrex, 2022-03-25:
Pangolin v0.5
OpenCV4.5.5 with CUDA
Eigen3.1.0
CUDA-11.4.15

jzhzhang, 2022-03-26:
@yCOrex I just tested the code with the following environment and meet the same error as yours.
```Ubuntu 18.04 LTS
Pangolin v0.5
OpenCV4.5.0 with CUDA
Eigen 3.3.0
CUDA-11.4
```
However, after upgrading the Pangolin version to `v0.6`, the error disappeared. I do not konw if it is related to some modifications in pangolin or some potential BUGs in the code.

Can you try to build your environment with Pangolin v6.0.  I will add more details to the README, if it works.

Tavoxluo, 2022-03-28:
I found Eigen3.3.0 not support cuda11.

jzhzhang, 2022-03-29:
Sorry for being misleading. I confirm that the version of eigen is 3.3.9

yCOrex, 2022-03-30:
Finally, after upgrading Pangolin version to `v0.6` , Eigen version to `3.3.9`, it `make` successfully
I think u need to emphasize the dependency on the corresponding version
At last, thx for patiently answering my questions

closed 2022-03-30 by yCOrex