jzhzhang/ROSEFusion #21: Frame could not be processed when running example

wuqs-net, 2022-03-30:
Hello, I'm using a GTX 1050 Ti (with CUDA 11.6) to run the program with the example data set. I have updated CMakeLists.txt accordingly and changed the PST path in the config file. But I still got the following output when running:

```
n:0
n:1
Frame could not be processed
n:2
Frame could not be processed
n:3
Frame could not be processed
```

The window displayed the frames correctly, but the output PLY does not contain any vertex.

wuqs-net, 2022-03-31:
Thanks for your response! My environment is

GPU: GTX 1050 Ti (compute capability 6.1, I have changed to this in CMakeLists.txt)
Ubuntu 20.04
CUDA 11.6
OpenCV 4.5.5
Eigen 3.3.9
Pangolin 0.6

wuqs-net, 2022-04-02:
Update:
I created my own docker image with CUDA 11.1.1 and the example ran successfully on it. CUDA 11.6 with other configurations unchanged did not work. The problem is probably related to CUDA version.

jzhzhang, 2022-04-02:
I also tested with CUDA 11.4, and it worked. I guess the problem may relate to the CUDA version or any other libs, which is also mentioned in [issue 20 ](https://github.com/jzhzhang/ROSEFusion/issues/20).

Anyway, I will mark this issue for future investigation.

hpf9017, 2022-04-15:
大佬，你好
我的环境，ubuntu 20.04  cuda11.4 opencv-4.5.0  ceres1.14
跑example.seq，但是结果还是Frame could not be processed

njk378390126, 2022-04-25:
> 大佬，你好 我的环境，ubuntu 20.04 cuda-11.4 opencv-4.5.0 ceres1.14 ，eigen3.3.7, 显卡rtx2060，算力7.5，都相应的改了 跑example.seq，但是结果还是Frame could not be processed

你好，我也遇到了同样的问题。我在3070显卡上不论是自己配置环境还是使用docker都有这个报错。但是切换到2080的机器上就没有这个问题了。问题应该在cuda那块。

njk378390126, 2022-04-26:
> 大佬，你好 我的环境，ubuntu 20.04 cuda-11.4 opencv-4.5.0 ceres1.14 ，eigen3.3.7, 显卡rtx2060，算力7.5，都相应的改了 跑example.seq，但是结果还是Frame could not be processed

在作者提醒下我解决了问题，最终发现3070的算力应该设成86。我看到你的eigen版本不对，切换成3.3.9会成功吗

closed 2022-06-15 by jzhzhang