jzhzhang/ROSEFusion #22: Frame could not be processed

hpf9017, 2022-04-15:
大佬，你好
我的环境，ubuntu 20.04 cuda-11.4 opencv-4.5.0 ceres1.14 ，eigen3.3.7, 显卡rtx2060，算力7.5，都相应的改了
跑example.seq，但是结果还是Frame could not be processed

fang-lai, 2022-07-16:
I also have the same problem

ubuntu 20.04 
rtx3060
cuda 11.7
opencv 4.5.5
eigen 3.3.9
sm_86

it mentioned that  "Frame could not be processed"

and docker image  in  [One drive](https://1drv.ms/u/s!AvuKnc9E9hmqhh_D_8pksC-cC2ma?e=K5DFnN)
no permission to download

jzhzhang, 2022-07-16:
@hpf9017 @fang-lai  Sorry for the late reply.  Please check the Opencv is built successfully with CUDA (turn on the ```opencv with CUDA``` option).  I also recommend you use the docker image if you are not sure your installation is correct.

jzhzhang, 2022-07-28:
If you have further questions, feel free to reopen the issue.

closed 2022-07-28 by jzhzhang