jzhzhang/ROSEFusion #23: OneDrive  rosefusion_docker.tar 文件下载下来无法解压,解压提示出错.

cfscn, 2022-06-10:
下载了很多次 都有该问题.

cfscn, 2022-06-10:
@jzhzhang 大佬,您看下原因呢?

jzhzhang, 2022-06-10:
你好，我们之前把docker镜像从google drive迁移到了One drive。过程中可能破坏了文件的完整。 目前的one drive里的镜像MD5为61a963894d79a77cd7a58eb37538be35，麻烦你确认一下不是下载的问题。 我们也会重新确认一下文件是否正确。

We have moved the docker image from google drive to one drive, and there may counter some mistakes while processing the image. The MD5 of image in one drive is 61a963894d79a77cd7a58eb37538be35. Please make sure the downloading is successfully finished. We will check whether the image is broken.

cfscn, 2022-06-10:
[image]
我对比了MD5 是正确的. 您看下是否文件完整被破坏了.

cfscn, 2022-06-10:
@jzhzhang  请问下 我现在去哪里能下载 这个镜像呢？

jzhzhang, 2022-06-10:
@cfscn  刚刚更新了one drive里的docker。 你可以重新在README里的链接里下载。 新的MD5码为2afeb0ac6793f650a72521655e4e3bbc

如果还有新的问题，欢迎随时联系我

cfscn, 2022-06-11:
好了 下载下来能用了

jzhzhang, 2022-06-11:
@cfscn  好的，感谢你的反馈

closed 2022-06-11 by jzhzhang