jzhzhang/ROSEFusion #24: Frame could not be processed when running example

wm17373510, 2022-09-16:
Hello, when I was running the program with the example data set, that Frame could not be processed. The window could display the frames correctly, but the output PlyFile does not contain any vertex.

My environment is as follow:
GPU: GTX 1080 (I have changed the sm_xx and compute_xx to fit my graphics card)
Ubuntu 20.04
CUDA 11.4
OpenCV 4.5.5
Eigen 3.3.9
Pangolin 0.6

jzhzhang, 2022-09-16:
Sorry for all the troubled with the environment installation. We have noticed some similar issues.  And we believe this problem mostly relates to the old version of testing environment , e.g ubuntu 16.04. (also some other minor tricks, like building the opencv with ENABLE_CUDA option)

A quick solution is to use the docker image, which is already been verified. And we are also planning to improve this repo with the newest version lib and more detailed instructions. This would be finished in the following October.

closed 2022-09-23 by wangm-buaa

manhha1402, 2022-10-31:
Hi @jzhzhang ,
I also have the same error when running the example on ubuntu 20.04, OpenCV built with `ENABLE_CUDA=True`. I checked the code and the pose estimation always return False in the step `particle_evaluation` with `count_search=0`.
Could you please check the error pls? When can you update the repo?
Thank you :)

jzhzhang, 2022-10-31:
We found that there are many reasons that could lead to this bug( including ENABLE_CUDA, compute capability, and the version of Eigen). We are re-implementing the code,  and hopefully,  it will be released in the next few days.

jzhzhang, 2022-11-02:
@manhha1402 @wangm-buaa We have tested our code in ubuntu 20.04 and it works as usual. We directly use the latest (libs)[https://github.com/jzhzhang/ROSEFusion/blob/main/README.md#tested-environment-2]. 

Please make sure you have correctly installed the environment:
- Check the ```ompute_xx,code=sm_xx``` and make sure it is valid for your GPU device.
- Build the opencv with ```-D WITH_CUDA=ON``` and make sure the ```-D CUDA_ARCH_BIN=x.x``` is valid for your GPU device.

manhha1402, 2022-11-02:
Thank you @jzhzhang . The problem is eigen version in ubuntu 20.04. I have to upgrade to eigen3.4.0 to make it work

jzhzhang, 2022-11-02:
@manhha1402  Glad to know you fix the issue. I will also mention the version of Eigen in FAQ.