jzhzhang/ROSEFusion #26: how to use FastCaMo-Synth ground truth trajectories

zhuisa, 2023-01-13:
请问，如何正确的使用参考相机轨迹数据？我使用如下的代码将轨迹数据转换成变换矩阵：

jzhzhang, 2023-01-14:
Hi, 我们提供的轨迹中 x, y, z 是世界坐标系下的相机position， 同时qx, qy, qz, qw 是相机朝向。 所以你可以先用旋转将相机坐标系下的点云转换到世界坐标系下， 再将点云平移到相机世界坐标系下的position

zhuisa, 2023-01-14:
@jzhzhang 我是这样做的，选择FastCaMo-synth-data(raw)的数据集office_0。根据相机内参和深度缩放因子，利用深度图计算相机坐标系下的点云模型，相机内参和深度缩放因子如下：
Camera.fx : 320.0
Camera.fy : 320.0  
Camera.cx : 319.5  
Camera.cy : 239.5  
DepthMapFactor: 1000.0
接着读取对应的gt文件，我认为gt的格式是x y z qx qy qz qw，然后计算旋转矩阵和平移矩阵得到变换矩阵：
T.pretranslate(t).rotate(q);
T.matrix();
最后，将相机坐标系下的点云乘以变换矩阵，但是得到的大模型还是混乱的，我不知道问题出现了哪里，所以请问可以展示一下代码，如何将根据参考轨迹得到完整的模型吗？很感谢您！

jzhzhang, 2023-01-14:
我手边并没有直接可以使用的代码。我的建议是 你可以输出你的转换矩阵T， 然后核对旋转矩阵部分和平移部分符合预期。你可以参考 关于转换矩阵组成的一个[例子](http://lavalle.pl/vr/node80.html)。

closed 2023-01-16 by jzhzhang