jzhzhang/ROSEFusion #28: Runtime on CPU

hannibal051, 2023-04-16:
Is the "particle evaluation" function running in real-time on a modern CPU possible?
BTW, I wanna inquire about the sampling (and evaluating) and evolving module in ASRO-DIO. Since you mentioned both CPU and GPU usage in the paper, is the aforementioned module running in real-time on a CPU possible?

jzhzhang, 2023-04-17:
Sorry for the late reply. 

The particle evaluation process is highly computationally expensive, involving numerous TSDF queries, making it difficult to achieve real-time performance using only the CPU.

No, the ASRO-DIO is tested on the device with both CPU and GPU. But it's slightly faster than the ROSEFusion, because of the pre-integration of IMU measurements.

The code of ASRO-DIO will be released in a few days, and you can refer to the code for more details.

hannibal051, 2023-04-17:
I see. From my perspective, for any optimization-based problem, if the Gaussian-Newton could achieve real-time performance, the random optimization should be faster.

hannibal051, 2023-04-25:
Dear Zhang:
I just checked the particles for transformation, which have 10240*20 + 3072*20 + 1024*20 particles. All x, y, and z conform to Gaussian distribution. I have two questions.
1. What's the correlation between these particles since q_trans has 60 members?
2. You limited the random optimization number of cycles to 20, so you only chose 20 among 60 q_trans, particle_index={0, 1+20, 2+40, ...}. How do you design the particle_index?

jzhzhang, 2023-04-25:
@hannibal051  
1) The q_trans are generated with different random seeds to have more diversity. q_trans[0] and q_trans[1] share the same meaning while with different random seeds.

2) The method skips some indexes just to use different q_trans. It's intended that some q_trans are not used. And you can change the index design to use other designs.

closed 2023-05-05 by hannibal051

zhangzscn, 2023-05-17:
When will the ASRO-DIO source code be released？