jzhzhang/ROSEFusion #30: Groundtruth trajectories for FastCaMo-Real.

Boyuan-Tian, 2023-08-25:
Hi Authors, 

Thanks for the great job!

Just want to know if there are ground truth trajectories provided for FastCaMo-Real?

I checked the one-drive for image sequences and downsampled ground truth meshes but haven't found trajectories.

Thanks!

jzhzhang, 2023-08-26:
We do not have the ground-truth trajectories for FastCaMo-Real. FastCaMo-real is composed of various large-scale scenes which are challenging to deploy a tracking system for each scene. Instead, we provide high-quality laser reconstructions to evaluate reconstruction results, such as completion and accuracy.

Boyuan-Tian, 2023-08-26:
Thanks for the comment!

So it seems like the FastCaMo-Real can only be used for end-to-end reconstruction evaluation instead of separately evaluating tracking and data integration right?

I was wondering to use Colmap or ROSEFusion to generate trajectories as a "pseudo ground-truth." It turns out that ROSEFusion is the only one that succeeds on fast motions such as FastCaMo-Real (Table 6) but slightly worse than typical methods on datasets with moderate motions (Table 3). Given this, do you have a rough idea that if the tracking poses from ROSEFusion on FastCaMo-Real are sufficient enough as a pseudo ground-truth?

jzhzhang, 2023-08-26:
Due to the lack of gt trajectory, it's intended that the FastCaMo-Real dataset be used for the reconstruction evaluation.

Yes, I think the tracking results of ROSEFusion can be reference trajectories to evaluate methods that perform way worse than ROSEFusion (with carefully selected scenes), just like Scannet for implicit SLAM (iMAP). Also I think you should mention this choice in your paper.

Boyuan-Tian, 2023-08-27:
Gotcha! Thanks!

closed 2023-08-27 by Boyuan-Tian

Boyuan-Tian, 2023-08-27:
Btw I decided to test other algorithms using the tracking trajectories generated by ROSEFusion, but it would be great if you accidentally have them somewhere?

jzhzhang, 2023-08-27:
Sorry. I would be happy to assist you with these results. However, I conducted this project during my master's program, and it has been 2 or 3 years since I last accessed these results. Retrieving these results might be challenging at this time.