jzhzhang/ROSEFusion #32: I want to ask some questions about the examples of reconstructed images in your paper.

followertryingtobeprogrammer, 2023-11-09:
[image]
The result shown in the figure seems to have no ceiling. May I ask if you extracted the point cloud from TSDF, reconstructed it into a mesh model, and then clipped it?

jzhzhang, 2023-11-09:
The shown figure is directly rendered with points cloud.  And yes, the ceiling is clipped for better visualization.

jzhzhang, 2023-11-10:
We use [open3d](http://www.open3d.org/) to generate the cylinders to represent the trajectory and save them as meshs. Then we use [cloudcompare](https://www.danielgm.net/cc/) to render both the trjactory mesh and scene point cloud.

closed 2023-11-30 by jzhzhang