jzhzhang/ROSEFusion #33: About Random Optimization Technique

dongho-Han, 2023-11-30:
Thanks for the nice work and implementation!
I am trying to run your code.
But, I got several questions below.

I understood Pose Optimization based on PFO with PST by original paper(about 17 pages). Especially, through Algorithm 1.
However, in the [arXiv's link](https://arxiv.org/abs/2105.05600), I can't find the supplement material(maybe supplement paper?), so I can't understnda the RescalePST in detail.
Can you share the supplement file?

Especially I'm curious about,
(1) Why the final shape of the PST is smoothen with previous step with [the condition previous_success & success](https://github.com/jzhzhang/ROSEFusion/blob/39f78878acd9de0ea2ad62a4f595fbcb437c652f/src/pose_estimation.cpp#L173)? it is not  directly mentioned in the original paper, only showing eq (18), not the condition. 

(2) I'm interested in the paper named [HOTrack(AAAI 2023)](https://pku-epic.github.io/HOtrack/) which also uses the random optimization technique as yours. But, as your paper says(in Implementation details chapter), with PST, 
'''
we store the original
PST corresponding to the unit sphere and move and rescale this original PST in each iteration with properly computed transformations
on the fly. 
'''
It also works on HOTrack. However, as the points are transformed(move and rescale) with ['update_seach_size' function](https://github.com/jzhzhang/ROSEFusion/blob/39f78878acd9de0ea2ad62a4f595fbcb437c652f/src/pose_estimation.cpp#L31C26-L31C26), it doesn't implemented as same as eq (16) & (17). I think the difference comes from only changing the transformation value('mean_transform' value), but I couldn't understand why the code is implemented like that. Can you kindly explain on those?

I wonder those two questions((1) & (2)) are written in supplement file. Can you give some help? Thanks in Advance.

jzhzhang, 2023-11-30:
Thank you for your interest in our project. The supplemental material can be found at [ACM website](https://dl.acm.org/doi/abs/10.1145/3450626.3459676).

Regarding your queries:
- Smooth Condition: Including isotropic sampling results in the smoothing operation is unnecessary. Hence, we have integrated the 'previous_success' parameter to circumvent such scenarios. The enhancement offered by this design is marginal; thus, we deemed it unnecessary to elaborate on this in the main paper.

- I'm not certain I fully understand your point.  'mean_transform' represents the difference between the new and old states, providing the direction for updates. 'Tsdf' measures the fitness. We also employ a value of 1e-3 and a 'scaling_coefficient' to ensure the validity of the search size. The implementation details are designed to ensure robust and reliable execution of the code.

Please let me know if you have more questions.

jzhzhang, 2023-12-01:
- （1） We add 1e-3 to avoid obtaining an extremely small denominator in the normalizing code. Additionally, this also ensures that Equation (17) does not yield a small search size, which might lead to the degeneration of PST scaling. Consequently, it is more practical to utilize the absolute vector for the addition operation. Otherwise, we would need to account for the sign of the `mean_transform`. There may be oversights in Equation 17, as it seems to suggest that a negative value could be added to 1e-3. I apologize for any confusion this may have caused.

- (2) Indeed, our code meticulously maintains a valid search range for teaching purposes. It's unlikely to precisely fit the search range through a purely mathematical approach. Therefore, we use the scaling_coefficient to ensure the search range's validity. Increasing the scaling_coefficient can enhance robustness while decreasing it can improve accuracy.

- (3) The term p(s_k*) represents the likelihood of the objective function. This likelihood increases when a good camera pose is sampled. However, for optimization purposes, the sample search range should be reduced when a good state is achieved, to ensure smooth convergence. The key is to find an effective cost function, in this case, the `energy`.

dongho-Han, 2023-12-02:
Thanks for the nice answer!
To summarize your answer in my understanding:
(i) making `s = mean_transform.abs() + 1e-3` is to get the valid search size in every iteration. Specially, `1e-3` is for evicting the extremely small `mean_transform` vector, and `mean_transform.abs()` is for ease to use the `1e-3`.
(ii) In eq (17), '1-p(s_k*)'  is used to make the fitness function value as large as possible(better solution). But, as in our case, the fitness function value means the `energy`, so we're trying to calculate `r_k^` vector proportional to the p(s_k*).

Then,
(1) Can you give any idea for 'a small search size, which might lead to the degeneration of PST scaling'? As i'm not good as mathematics, I couldn't get the idea for the degeneration of PST scaling and what will be occured. 
**I thought that making search size small as possible(nearly zero-size sphere), we can say that we got the optimal solution.**

(2) Can you tell me 
'Indeed, our code meticulously maintains a valid search range for teaching purposes. It's unlikely to precisely fit the search range through a purely mathematical approach. Therefore, we use the scaling_coefficient to ensure the search range's validity.'
with more detail??

**I understood that `scaling_coefficient` can enhance robustness or acc, but couldn't understand**
(i) the meaning of `teaching purposes`

(ii) Could you tell any examples that mathematical approach can't reach the optimal search range?

(iii) how can the `scaling_coefficient` ensure the search range's validity? what is the meaning of search range's validity? Maybe it could be related to the valid search range in the first sentence. I think the valid search range is almost 0, but it's incorrect from your reply.

Sorry for the dumb questions..

jzhzhang, 2023-12-02:
If the search range of one dimension is really small, the next averaged sampled state will exhibit negligible updates in that dimension. This will further causes small seach range in the next sampling. In this case, this dimension is ignored in the whole optiomization process.

The random optimization is one type of evolution strategy. Typically, the objective function and the search range are decoupled. You refer to evolution strategy methods for more details (https://en.wikipedia.org/wiki/Evolution_strategy, https://cma-es.github.io/ )

dongho-Han, 2023-12-02:
Thanks for the answer! I really appreciate your efforts.

closed 2023-12-02 by dongho-Han