jzhzhang/ROSEFusion #38: About the reconstruct point cloud distance metric

bowieshi, 2024-10-02:
Hi. I have run rosefusion in my own dataset. I find the reconstruct point cloud has very large (x, y, z), which seems not align with the estimated camera pose metric. However, even scale the point cloud coordinate, the camera pose seems not in the right place. The first camera has pose (0, 0, 0, 0, 0, 0, 1). So I want to inquire what is the coordinate convention amd metric of the reconstructed scene and estimated camera pose. How to align them in the same metric space. Thank you!

[image]

here is a clip of my reconstructed point cloud ply file:
```
ply
format ascii 1.0
element vertex 425266
property float x
property float y
property float z
property float nx
property float ny
property float nz
property uchar red
property uchar green
property uchar blue
end_header
227.043 3790 2790 0.897481 -0.155614 0.412689 109 107 110
221.28 3810 2810 0.817679 -0.572926 -0.0561833 109 107 109
228.772 3830 2810 0.895167 0.324596 0.305472 109 108 104
190 3810 2832.09 -0.726066 -0.217654 0.652269 108 108 108
218.279 3810 2830 0.601139 -0.0677041 0.796271 110 109 106
218.279 3810 2836.25 0.601139 -0.0677041 0.796271 110 109 106
150 3830 2849.33 0.267826 -0.355996 0.895286 107 110 109
190 3830 2838.76 -0.430508 -0.814946 0.387976 107 107 107
226.055 3830 2830 0.43364 0.211501 0.875913 110 109 105
226.055 3830 2837.95 0.43364 0.211501 0.875913 110 109 105
148.768 3830 2850 0.648259 0.0120811 0.761324 108 111 110
148.768 3830 2865.98 0.648259 0.0120811 0.761324 108 111 110
153.12 3830 2850 -0.297894 -0.354116 0.886488 108 111 110
```
Here is the estimated camera pose file:
```
0 0 0 0 0 0 0 1
1 -0.00770302 5.08576e-05 0.00266247 -0.00984742 0.00361598 0.00115856 0.999944
2 -0.00500228 0.000558167 0.00251346 -0.0345661 0.017153 -0.00627817 0.999235
3 -0.00237271 0.00217353 0.000952712 -0.0312629 0.0205179 -0.0189311 0.999121
4 -0.00568282 0.00653649 0.000924178 -0.0433483 0.0110819 -0.0186088 0.998825
5 -0.00830067 0.00191484 0.00265268 -0.0399927 0.00893834 -0.0253685 0.998838
6 -0.00517633 0.00319866 5.90482e-06 -0.0331452 0.0065474 -0.0302181 0.998972
7 -0.0086429 0.00800957 -0.00189315 -0.038435 -0.00208478 -0.0381654 0.99853
8 -0.010853 0.00174917 -0.00223585 -0.0432972 -0.0031492 -0.0385713 0.998312
9 -0.0125234 -0.00454584 -0.000587218 -0.0405253 -0.00200062 -0.0436311 0.998223
10 -0.0146637 -0.0195738 0.00527937 -0.037216 -0.0113672 -0.0402919 0.99843
11 -0.0134601 -0.029709 0.0301963 -0.0298153 -0.0170401 -0.0516046 0.998077
12 -0.00569468 -0.0435402 0.0572829 -0.017359 -0.0247136 -0.0617147 0.997637
13 0.00274669 -0.0648414 0.0648085 -0.029405 -0.00904897 -0.0777481 0.996498
14 0.00618364 -0.0850539 0.0656771 -0.0319983 0.00247609 -0.0893872 0.99548
15 -0.0063644 -0.121949 0.0588209 -0.0351541 -0.00383882 -0.08391 0.995846
16 -0.0205114 -0.17462 0.0518017 -0.0219607 -0.016303 -0.0911924 0.995458
17 -0.0311161 -0.215621 0.0437636 -0.00431829 -0.0252364 -0.0928961 0.995347
18 -0.0419745 -0.24498 0.0415615 0.00543834 -0.0258312 -0.0933107 0.995287
19 -0.0408249 -0.284035 0.0355478 0.0121421 -0.0208749 -0.0935184 0.995325
20 -0.0366889 -0.326069 0.0262346 0.0129234 -0.0171156 -0.0985158 0.994904
21 -0.0317068 -0.359519 0.0203467 0.0165733 -0.0127823 -0.101472 0.994618
22 -0.0259384 -0.394391 0.0140853 0.00515829 -0.00556288 -0.0994558 0.995013
23 -0.0271315 -0.430521 0.015088 0.00484731 0.00121597 -0.0885752 0.996057
24 -0.0233585 -0.454188 0.0168206 0.00586718 0.0160211 -0.0778761 0.996817
25 -0.0157577 -0.468349 0.0200675 0.00740385 0.0216976 -0.0771319 0.996757
26 -0.0115286 -0.490267 0.0235701 0.0072794 0.0188679 -0.0787593 0.996689
27 -0.00975674 -0.513074 0.0290574 0.00880237 0.00835395 -0.0796213 0.996751
```
I have set the `depth_ratio: 1000.0` in `seq_generation.yaml`

jzhzhang, 2024-10-02:
Please check that your depth camera settings align with the scale settings (Kinect represents 1000 as one meter, and Realsense uses 5000 to represent one meter).

As for the origin of the camera pose, I’m also confused. Could you provide the results for just a few frames, like 5 to 10?

bowieshi, 2024-10-02:
@jzhzhang Thank you for your reply. I am using Orbbec camera. My depth png file has shape of (480, 640). The Orbbec represents 1000 as one meter (like depth information likes "365", "1521", "1986", which is 0.365m, 1.521m, 1.986m). After running `./seq_gen` to get `.seq` file and then run ROSEFusion main function, I can get a `.ply` file. But I found the coordinate of the reconstructed point cloud is very large (the x, y, z are typically "227.043" "3790" "2790", it is very large and far from the original point, like the image above).
But the output camera pose in `.txt` file is relatively reasonable. It has the first camera pose is transl (0, 0, 0), orientation (0, 0, 0, 1). And last camera pose is transl (-3.74, -0.14, 0.44). It looks well within the room record traj. 
My understanding that the point cloud should be in the same metric space as the output camera pose. And the first camera should in the origin of the system. But it seems the point cloud does not match with the camera. This is where I got confused.
Also, can I inquire what is the camera pose definition convention? (world to camera or camera to world) And the camera view coordinate system axis direction.
Thank you for your help!!

bowieshi, 2024-10-02:
[image]

jzhzhang, 2024-10-02:
The reconstruction scale should be 1000 per meter, so the x, y, and z coordinates are represented in millimeters.

For the coordinate system, the original camera pose is placed at the center of the volume (with an offset if configured). Refer to your data configuration file for detailed volume parameters.

Camera poses are defined relative to the world coordinate system.

If you still feel confused, please let me know.

bowieshi, 2024-10-02:
@jzhzhang , thank you for your help. I solved my problem. This origin of reconstructed points cloud is located as (voxel_x / 2.0, voxel_y / 2.0, voxel_z / 2.0), which is defined at `data.yaml`. This voxel should also be scaled by depth_ratio and voxel_size. The camera view coordinate frame is -z forwards, y upwards, x right.

bowieshi, 2024-10-08:
@jzhzhang Hi, I still find the pose definition unclear. Can I ask the pose is defined as projection from world to camera or from camera to world? Also, what is the camera view frame axis direction? (like -z forwards, y upwards, x right defined in blender)

closed 2024-10-08 by bowieshi