jzhzhang/ROSEFusion #39: Masking human for indoor camera localizatrion

bowieshi, 2024-10-27:
Hi. If I have a RGBD video where a moving human is present. Is it possible for me to mask out the human by some segmentation model and then use ROSEFusion to estimate camera moving trajectory?

jzhzhang, 2024-10-27:
ROSEFusion is a frame-to-model pipeline that requires depth information to compute the relative pose for the reconstruction model.

Thus, it depends on the remaining depth pixels after masking out the human. If a substantial number of depth pixels remain, ROSEFusion should function effectively.

bowieshi, 2024-10-30:
@jzhzhang  Thank you for your quick reply! Could you give me some hints where to modify your code in order to add a human mask? Thank you again!

jzhzhang, 2024-10-31:
We do not have such an API to mask out pixels.  ROSEFusion will read the depth map, and directly perform tracking. The related code can be found in [here](https://github.com/jzhzhang/ROSEFusion/blob/8cf122b9c7f7403efe90dca3b0af0ebfec10b26d/src/seq_gen.cpp#L65). You can directly add masked depth map, then run ROSEFusion.

closed 2024-12-02 by bowieshi